Limine-Bootloader/Limine #492: Chainloading from other devices not working after 9.0.1

glics, 2025-04-12:
Running Limine 9.0.1 up to 9.2.2, on a x64 UEFI system (Thinkpad T14 Gen1 AMD, AMD Ryzen 5 PRO 4650U).
Secure Boot is disabled.
Here's the relevant configuration:

```
/Windows
    protocol: efi_chainload
    path: uuid(FE8D3BBB-9AB8-433C-A3FB-42FA7D23DEA2):/EFI/Microsoft/Boot/bootmgfw.efi
```

Trying to boot this entry results in a reboot of the machine which starts back into Limine;
Works fine on 9.0.0.

Here's what I have attempted:
- Trying to boot `/EFI/Boot/bootx64.efi` results in the same reboot;
- Trying to boot any other EFI file results in the same reboot, including an EFI Shell and Limine's own .efi copied to the other ESP;
- Trying to boot using the `hdd()` command results in the same reboot;
- Copying `bootmgfw.efi` to the same ESP as Limine is installed on, and booting it as `boot():/bootmgfw.efi` results in a successful chainload (but won't boot windows without copying other files from Windows's ESP);
- Trying to boot any .efi file from the same ESP using `boot()` works;
- Trying to purposefully give a wrong UUID or path results in a panic message as expected (Failed to open image with path...), without rebooting;
- Using `efi` instead of `efi_chainload` and `guid()` instead of `uuid` makes no difference.

I have another machine (also UEFI x64, a desktop) on which I wasn't able to reproduce the issue, it boots from other partitions just fine.
Assuming this is a setup issue, I'm not sure how to further diagnose the issue unfortunately

Narzaru, 2025-07-13:
I have a similar problem. Windows does not load with the same error `Failed to open image with path..`.

Here is my configuration.
```
arch% efibootmgr
BootCurrent: 0001
Timeout: 1 seconds
BootOrder: 0001,0000,0005,0006,0007
Boot0000* Windows Boot Manager	HD(1,GPT,c4eac268-ed4e-4aab-b785-e8111b67070a,0x800,0x32000)/\EFI\Microsoft\Boot\bootmgfw.efi57494e444f5753000100000088000000780000004200430044004f0042004a004500430054003d007b00390064006500610038003600320063002d0035006300640064002d0034006500370030002d0061006300630031002d006600330032006200330034003400640034003700390035007d00000000000100000010000000040000007fff0400
Boot0001* Arch Linux Limine Bootloader	HD(1,GPT,65000f7d-1617-4628-8fa2-39a096010fee,0x800,0x200000)/\EFI\limine\BOOTX64.EFI
Boot0005* UEFI:CD/DVD Drive	BBS(129,,0x0)
Boot0006* UEFI:Removable Device	BBS(130,,0x0)
Boot0007* UEFI:Network Device	BBS(131,,0x0)
arch% lsblk -f -o +PARTUUID           
NAME   FSTYPE FSVER LABEL      UUID                                 FSAVAIL FSUSE% MOUNTPOINTS PARTUUID
sda                                                                                            
└─sda1 ntfs         New Volume 86202A4A202A4213                                                4f7ec0a7-92ae-4fc3-ae55-167e191cfe4e
sdb                                                                                            
├─sdb1 vfat   FAT32            008E-9DE6                                                       c4eac268-ed4e-4aab-b785-e8111b67070a
├─sdb2                                                                                         e0e03e1a-d0b5-41e0-bea6-7c377052ac73
├─sdb3 ntfs                    AC249B4B249B1802                                                cb0f736d-b59c-4cd4-ab49-7e59b21b2788
└─sdb4 ntfs                    C276C16276C157B5                                                186560b6-cc9a-4b80-bd35-ed644a3929b2
sdc                                                                                            
├─sdc1 vfat   FAT32            B775-623E                             832.8M    19% /boot       65000f7d-1617-4628-8fa2-39a096010fee
└─sdc2 ext4   1.0              cc029104-fc3a-41b1-bd05-552ffe3a6d0a  394.4G     5% /           ddba531d-2ae4-46b4-bcc5-40e278166f61
zram0  swap   1     zram0      b5e4eae6-bea2-4eca-9e26-47bb4de99cb1                [SWAP]      
arch% cat /boot/EFI/limine/limine.conf
timeout: 5

/Arch Linux (linux)
    protocol: linux
    path: boot():/vmlinuz-linux
    cmdline: root=PARTUUID=ddba531d-2ae4-46b4-bcc5-40e278166f61 zswap.enabled=0 rw rootfstype=ext4
    module_path: boot():/initramfs-linux.img

/Arch Linux (linux-fallback)
    protocol: linux
    path: boot():/vmlinuz-linux
    cmdline: root=PARTUUID=ddba531d-2ae4-46b4-bcc5-40e278166f61 zswap.enabled=0 rw rootfstype=ext4
    module_path: boot():/initramfs-linux-fallback.img

/Windows 11 (Windows)
    protocol: efi
    path: uuid(c4eac268-ed4e-4aab-b785-e8111b67070a):/EFI/Microsoft/Boot/bootmgfw.efi
```

qookei, 2025-08-31:
Development has moved to Codeberg: https://codeberg.org/Limine/Limine. Please reopen this issue there.

closed 2025-08-31 by qookei