Limine-Bootloader/Limine #495: Limine bootloader screen garbled at boot on 2010 Macbook Pro (Nvidia 320m GPU)

kouan9ka, 2025-04-12:
Hardware:
2010 Macbook Pro 13" (MacbookPro 7,1)
CPU: intel core2duo P8600
GPU: nvidia geforce 320m
8GB RAM

Software:
OS: CachyOS
Limine version: 9.2.2

[image]

Symptom:
At boot, limine screen is completely garbled. operation still works, and can select boot entry with arrow keys as expected.
OS will boot, and all operation and display is fine once past limine bootloader screen.

[image]

braewoods, 2025-04-13:
Are any other bootloaders able to work properly? This might have something to do with older Macs having an older `EFI` based on `1.x` or so. I believe there are two graphical standards for `EFI` so it might just be a matter of supporting both in some sense. `UGA` vs `GOP` according to `osdev`.

https://wiki.osdev.org/GOP

LordKeravnos, 2025-04-13:
is it even EFI in the first place? I don't know a lot about macs but I've never seen efi on a core2 system. i know apple like to have more unique firmware, so if it's trying to use the dGPU for the frame buffer, could there be something it doesn't like there? just spitballing out of curiosity.

braewoods, 2025-04-13:
It boots using EFI but you're right in the sense that it's fairly non-standard. It might help to look at a bootloader like `Clover` or `Refind` to find out if they're doing something special for Apple hardware.

LordKeravnos, 2025-04-13:
interesting, thank you. I guess all we can do for now is wait for OP to run some tests for us to pull apart. I've never really used much else than Grub or Limine, and the latter mostly because I got annoyed with how clunky the former feels. godspeed my friends, I hope OPs issue can get fixed soon.

mintsuki, 2025-04-13:
Hello, could you try adding `interface_resolution: 800x600` or `1024x768` to your config file to see if that mitigates the issue? Additionally, it is quite unlikely, but it could be that 9.2.3 fixes the issue since it fixes a NULL pointer dereference under certain circumstances with certain wallpaper images that was present in 9.2.2 and 9.2.1.

kouan9ka, 2025-04-13:
I added interface_resolution to the config file at 1200x800 (displays native res) and 800x600, and there was no change or difference (still garbled).

grub i believe works/displays correctly on this machine (the USB ISO i used to install the OS booted with grub, and the grub menu was no problem).

If there is anything else i can try, or any other info or data that would be helpful, let me know.

LordKeravnos, 2025-04-13:
Have you updated Limine since 9.2.3 came out? make sure you copy over the new BOOTX64.EFI file if your pkgmanager doesn't have a hook for it.

kouan9ka, 2025-04-13:
ah, didnt realize 9.2.3 was out. i updated just now (pacman updates it correctly as far as i can tell).
Still same issue occurs with 9.2.3, both with and without the 'interface_resolution' argument in the config file.

hustlerone, 2025-06-09:
FYI this also happens on the steam deck if you set the resolution to 1280x800

closed 2025-08-31 by qookei