Limine-Bootloader/Limine #497: Protocol documentation for "Executable Address feature" is outdated

Magyar57, 2025-04-13:
Hi,

As the title says, the [aforementionned documentation](https://github.com/limine-bootloader/limine/blob/v9.x/PROTOCOL.md#executable-address-feature) is outdated: it misses the mention of the name of the feature/struct having changed after API revision >= 2

Shouldn't the documentation mention the difference in between the API versions ?

In `limine.h`:
```c
#if LIMINE_API_REVISION >= 2
#  define LIMINE_EXECUTABLE_ADDRESS_REQUEST { LIMINE_COMMON_MAGIC, 0x71ba76863cc55f63, 0xb2644a48c516a487 }
#else
#  define LIMINE_KERNEL_ADDRESS_REQUEST { LIMINE_COMMON_MAGIC, 0x71ba76863cc55f63, 0xb2644a48c516a487 }
#endif

// and so on for the structure too
```

mintsuki, 2025-04-13:
Hi,

The documentation documents only the latest API revision as anything before the latest is considered deprecated.

closed 2025-04-13 by mintsuki

Magyar57, 2025-04-14:
Oh okay. It seems that I'm misunderstanding how setting the base revision works.

In my code, I am setting `static volatile LIMINE_BASE_REVISION(3)`, after including `limine.h`. But this doesn't define LIMINE_API_REVISION to 3, it keeps the default 0 ; am I supposed to `#define LIMINE_API_REVISION 3` manually before including the header ? It's not what the [limine-c-template](https://github.com/limine-bootloader/limine-c-template/blob/trunk/kernel/src/main.c) shows, so I'm a bit confused

mintsuki, 2025-05-05:
They are different things. The base revision changes the binary interface with the bootloader and needs bootloader support, while the API revision doesn't require bootloader support and binaries made will be compatible with older bootloader versions, as the API revision only really changes data types, variable names, or other somewhat "cosmetic" things in the programming interface.

Magyar57, 2025-05-05:
Thank you for the explanation. What I don't understand is, how do I make sure that when I include limine.h, it uses the LIMINE_API_REVISION that I want (3, and default is 0) ? I'm supposed to #define it before including the header ?