Limine-Bootloader/Limine #499: Documentation for FreeBSD?

chexo3, 2025-04-30:
Hello! I see @mintsuki has been [authoring ports](https://www.freshports.org/sysutils/limine/) for Limine for FreeBSD. Is there documentation somewhere that I'm missing for how to set up Limine to boot FreeBSD? Can Limine boot FreeBSD directly or does it need to chainload FreeBSD's loader?

I ask because I am interested in installing Limine on my tinkering laptop to boot FreeBSD, and I'm not certain where to start.

kovmir, 2025-05-03:
1. [Limine cannot boot FreeBSD directly](https://github.com/limine-bootloader/limine/blob/82d2d4253e105e3c938396bf85c3da568f1ef73c/README.md?plain=1#L38C1-L43C15).
2. Limine can chainload UEFI bootloaders. Meaning your computer would start Limine, Limine would start FreeBSD bootloader, and FreeBSD bootloader would start FreeBSD operating system. This is pointless unless you have a dualboot system.

chexo3, 2025-05-03:
> 1. [Limine cannot boot FreeBSD directly](https://github.com/limine-bootloader/limine/blob/82d2d4253e105e3c938396bf85c3da568f1ef73c/README.md?plain=1#L38C1-L43C15).

Just because there’s not a “FreeBSD” protocol listed doesn’t mean it’s necessarily unsupported. Limine and Multiboot 1/2 both have a bunch of parameters related to placing a kernel and its modules in memory. I have no way of knowing if this could boot FreeBSD though and that’s why I’m asking.

> 2. Limine can chainload UEFI bootloaders. Meaning your computer would start Limine, Limine would start FreeBSD bootloader, and FreeBSD bootloader would start FreeBSD operating system. This is pointless unless you have a dualboot system.

Correct!

kovmir, 2025-05-03:
>Just because there’s not a “FreeBSD” protocol listed doesn’t mean it’s necessarily unsupported. 

"unsupported" is exactly what it means. No explicit support.

>Limine and Multiboot 1/2 both have a bunch of parameters related to placing a kernel and its modules in memory

FreeBSD does not support neither Limine nor Multiboot protocols.

chexo3, 2025-05-03:
> > Just because there’s not a “FreeBSD” protocol listed doesn’t mean it’s necessarily unsupported.
> 
> "unsupported" is exactly what it means. No explicit support.

Okay, fine, what I’m asking is whether there’s any way of making it work.

> > Limine and Multiboot 1/2 both have a bunch of parameters related to placing a kernel and its modules in memory
> 
> FreeBSD does not support neither Limine nor Multiboot protocols.

And that means it will never work?

kovmir, 2025-05-04:
> what I’m asking is whether there’s any way of making it work.

No, there is not.

> And that means it will never work?

Yes, it will not work. There is no overlap in supported boot methods across FreeBSD and Limine.

chexo3, 2025-05-04:
Okay, I’m just going to come out and say it at this point.

I don’t know what I’m talking about, but I don’t think you do either.

You have no commits to Limine (see attached screenshot) and I can’t find any mention of you searching the Matrix room, searching the entire Limine GitHub organization, I see your name come up in two other issues and neither of them lead me to believe you know what you’re talking about.

I’d like to be proven wrong, but frankly you seem like the type that sits on e.g. StackOverflow searching for an excuse to close posts over minor issues rather than providing help.

It’s a shame, because it seems like you do write some cool software, but none of it relates to this so I don’t know why you’re here.

[image]

kovmir, 2025-05-05:
> I don’t know why you’re here.

1. I *use* Limine, so I participate a little in the usage and new feature discussions. I hate all the other Linux bootloaders, especially their bloat and insane config file syntax, and I do not want Limine to face the same fate.
2. I *package* Limine.

> ... neither of them lead me to believe you know what you’re talking about.

Fine with me. I will not reply to your issues.

---

Simply put, Limine cannot boot FreeBSD (only chainload):

* Limine (the bootloader) does not support FreeBSD boot protocol.
* FreeBSD does not support either Limine (the boot protocol) or Multiboot 1/2.

chexo3, 2025-05-06:
@mintsuki thank you for clarifying. Sorry for getting heated, I was just frustrated because I’ve had situations before where people have been more interested in finding reasons to close discussions or flex some superiority rather than be helpful, and this felt somewhat like that.