Limine-Bootloader/Limine #500: rotate boot menu

cristids, 2025-05-08:
Hi,

I own a GPD Pocket 4 that has a strange screen rotated at boot. I can switch the text mode using fbcon but all the boot menus including grub, systemd are rotated 90 deg ccw. I was wondering if possible to rotate limine boot menu in some way.
This could probably benefit other people and devices who have the screen rotated at boot either hardware or because their screens are kept in portrait mode.

Thanks.

mintsuki, 2025-05-29:
Hi, sorry for the somewhat late reply.

This is something that I'd gladly add to the project, but it needs support from Flanterm first and foremost: https://codeberg.org/mintsuki/flanterm

I'll put a pin in it as something to implement over there, then once that's done I'll add a config option to rotate the interface.

qookei, 2025-08-31:
Development has moved to Codeberg: https://codeberg.org/Limine/Limine. Please reopen this issue there.