Limine-Bootloader/Limine #501: Defer freeing of the memory map entry `EfiBootServicesData` to be able to use the BGRT image

andIsome, 2025-05-17:
You can't display the vendor logo during kernel initialization. This image is located in the `EfiBootServicesData` mem-map entry (pointed to by the BGRT ACPI table) and is thus inaccessible when limine frees it.

This should be as easy of a fix as changing the type of mem-map entry limine passes the kernel: `MEMMAP_USABLE` -> `MEMMAP_BOOTLOADER_RECLAIMABLE` [here](https://github.com/limine-bootloader/limine/blob/5ff3bfa4348c08d0bdf6ce529337c7ff914a9098/common/mm/pmm.s2.c#L483C24-L483C25)

One could solve this issue in other ways, but this seems the most straight forward. The only drawback is that the kernel has less MAP_USABLE available, but the `EfiBootServicesData` doesn't use too much memory, right? I mean its will reclaimable anyway.

I checked on my linux machine and it appears that GRUB doesn't free the the `EfiBootServicesData` as linux has access to it (I can check it out at `/sys/firmware/acpi/bgrt/`)

Further info:
[ACPI spec](https://uefi.org/sites/default/files/resources/ACPI_5_1_Errata_B.PDF) mentions it on page 188.

mintsuki, 2025-05-17:
Hello, thanks for opening the issue!

I was not aware that ACPI tables could be in `EfiBootServicesData` regions until now. As a matter of fact it seems like the BGRT is the only exception to this rule, by a quick CTRL-C of the ACPI spec.

That being the case, I believe it would be better to change the memory range of the BGRT in the memory map to `MEMMAP_ACPI_RECLAIMABLE` instead of not freeing `EfiBootServices*` at all.

andIsome, 2025-05-17:
Well it seems I'm a bit to unfamiliar with the limine mapping stuff to modify it, in the way you suggested. But essentially one has to add

```c
struct bgrt {
    struct sdt header;
    uint16_t version;
    uint8_t  status;
    uint8_t  img_type;
    uint64_t img_address;
    uint32_t img_xoffset;
    uint32_t img_yoffset;
} __attribute__((packed));

struct bmp_header {
	uint16_t id; // magic number: 0x4d42
	uint32_t size;
} __attribute__((packed));
```
then use the (located at the beginning of `img_address`) `struct bmp_header` -> `size` to update the memory range's type (`img_address`; `img_address` + `size`)

[Linux reference implementation](https://elixir.bootlin.com/linux/v6.14.6/source/drivers/firmware/efi/efi-bgrt.c#L82)

mintsuki, 2025-05-18:
Can you please try to see if 5f458ea3 works for you? I went ahead and decided to take the safe option of making boot services regions bootloader reclaimable as you first suggested.

andIsome, 2025-05-18:
I can confirm that its located within
```
Start Addr: 0x7E8E1000 | Length: 0x00020D000 | Type: MEMMAP_BOOTLOADER_RECLAIMABLE
[...]
bgrt: phys: 0x7E8E6018, off: 415, 355
```
and i can display something that looks like a image

[image]

andIsome, 2025-05-18:
And it seems to works on hardware

[image]

closed 2025-05-19 by mintsuki