Limine-Bootloader/Limine #502: How to get usable virtual memory ranges?

ChocolateLoverRaj, 2025-05-23:
I know there is a Limine request which responds with the physical memory regions, and tells you which ones are usable. How are you supposed to get usable _virtual_ memory regions? One use case is mapping ACPI tables. Right now, in my kernel I traverse page tables to find unused ranges. However, this seems like a waste since Limine is setting up the page tables and then my kernel has to traverse everything again.

Am I missing something? Or could it be a new Limine feature that gives you usable virtual address ranges?

DeanoBurrito, 2025-05-23:
The layout of virtual memory is described under https://github.com/limine-bootloader/limine/blob/trunk/PROTOCOL.md#entry-memory-layout. If it's not described here, it's free or non-canonical.

ChocolateLoverRaj, 2025-05-23:
So @DeanoBurrito will this work?:
- Get the HHDM
- Use Memory Map Feature to get all 
    Usable
    Bootloader reclaimable
    Executable and modules
    Framebuffer
- Add HHDM to each physical region above to get the used virtual addresses
- Everything else is free?

If I do this is it guaranteed to not mess with pointers in my kernel and pointers in Limine responses? Are all pointers in Limine responses guaranteed to use HHDM?

ChocolateLoverRaj, 2025-05-24:
> Some general advice here is to move to your own page tables, and address space allocator

How to do this? How do you go backwards, finding the virtual address ranges based on the physical regions?

And even if I do create new page tables, I'll still need to keep the virtual addressesses for the kernel and Limine requests and responded the same.

> `All responses and associated data structures are placed in bootloader-reclaimable memory regions.`

But that's physical memory. I don't see how it's possible to get the virtual range for this, besides just adding the HHDM offset (but from what I read there is no guarantee that this is how it was set up).

ChocolateLoverRaj, 2025-05-25:
I managed to implement creating new page tables, mapping the HHDM stuff and the executable. But I have to use 4KiB pages to do this, because the physical start is 34111488 and the virtual start is 18446744071562067968. They are both aligned to 4KiB but not 2MiB or 1GiB. This is very inconvenient. Also, I don't see a way of getting the size of the executable in memory (which can be more than the size of the ELF, right?). So right now, I either have to guess (like assuming size in memory is <64MiB), or map the full 2GiB using 4KiB pages (which takes SUPER long).

ChocolateLoverRaj, 2025-05-25:
In the end I just ended up reusing the L3 page table from Limine for the last 512 GiB.

DeanoBurrito, 2025-05-26:
You can get the physical base of the kernel from a limine feature (this memory is contiguous so you only need the base). You can use some symbols in your kernel's linker script to get the sizes of the various sections or the whole kernel. That'll give the size at runtime, and the virtual base. That should be everything you need to make your own page tables :)

You can map the hhdm with larger mappings, but yeah the kernel will need 4K anyway due to the alignment of the segment headers.

As an aside, there's an osdev discord run by mintsuki (many limine contributors are there as well, as am I) you might be interested in joining, you'll get more timely responses. https://discord.gg/RnCtsqD

mintsuki, 2025-05-29:
Closing this as out of scope.