Limine-Bootloader/Limine #508: No boot():/ only in uefi.

Barteeq123, 2025-06-18:
I want to boot a elf32 file with a Multiboot2 header. So in the limine.conf file i use `bios():/` path to locate the `main.bin` file and this works well with UEFI but for some reason `bios():/` doesn't exist under BIOS legacy boot which is specified in `CONFIG.md`, so how am i supposed to find the `main.bin` file that i need to boot when just doing `/main.bin` or `/boot/main.bin` doesn't work?

mintsuki, 2025-06-20:
`bios():/` is not a thing. There is `boot():/`, and that works on both UEFI and BIOS.

closed 2025-06-20 by mintsuki

Barteeq123, 2025-06-21:
The `bios():/` thing was just a typo in my config file i used `boot():/` and under legacy bios it says that it couldn't find the file that i'm trying to find using `boot():/`, everything works under UEFI but that will make sense because it's even said in the file that explains how to make configs that `boot():/` is NOT available under legacy bios, but i can't find my file without it.

mintsuki, 2025-06-21:
What the `CONFIG.md` file means is that `boot():/` is not available if you use the SMBIOS method of providing a configuration file. Of course given that you're the second person the wording in the file misleads, then that wording should be changed.

If you don't use the SMBIOS method of giving a config to Limine (and you likely do not), `boot():/` will work just fine on legacy BIOS, and it will be the partition on which the `limine.conf` file is located. You can figure out the path from there.