Limine-Bootloader/Limine #512: Black screen output, also freeze PC on chainload EFI secure boot by shim

vuongtuha, 2025-06-24:
OS: fedora 42, shim and mmx copypasta from standard grub2 pkg
Boot chain: shimx64 (mmx64) → limine_fake_grubx64.efi → UKI (systemd-stub)

to simplify reproduce, just change UKI to refind-bootloader.efi.signed
[  BOOTX64.EFI → grubx64.efi → refind/grubx64.efi ]
UEFI settings: CSM disabled, secure boot enabled standard Windows mode, all vendor key stayed, SATA mode AHCI, 1st boot value is shimx64.efi 
Limine: version 9.3.3, idk why but 9.3.4 gave me UNVERSIONED
.sbat:```
 sbat,1,SBAT Version,sbat,1,https://github.com/rhboot/shim/blob/main/SBAT.md
limine,1,Mintski,limine,9.3.0,https://www.github.com/limine-bootloader
```
limine.conf:
```
/refind
   comment: boot pls dont block
   protocol: efi
   path: boot():/EFI/refind/grubx64.efi
```
key signed is pkcs11 standard. In case limine, origin BOOTX64.efi added sbat.csv 1st, sign by pesign. 
What happened in boot process:
- when disable secure boot, limine can finish boot chain, normal boot into UKI with all modules dkms
- reboot into uefi settings, change to enable secure boot, and reboot to shim
- enroll my MOK key .cer, which used for signed limine, then reboot again
- Limine screen appeared normally, can pick/edit boot value
- But after select to boot to "refind" opt, screen turn black, keyboard numlock wont change, single press power button wont work. So only choice is hard force PSU power button.
Had to note about something i've played on secure boot, bcoz i'm just bored:
Sbctl to sign limine, and enroll my custom keys. This setup can work very well on both UKI mode and separate vmlinuz+initrd mode. In this setup, your bootloader works well, even with signed modules.
Shim mode seem to transform your bootloader into preload.efi more. Forgive me, maybe I had no idea wtf i just have typed, but if there is any context need to be more clear, or somewhat was my mistake in process, just point me out.

vuongtuha, 2025-06-24:
```
/efi/EFI/
├── BOOT
│   ├── BOOTX64.EFI
│   ├── grubx64.efi
│   ├── limine.conf
│   └── mmx64.efi
├── refind
│   ├── BOOT.CSV
│   ├── grubx64.efi
│   ├── icons
│   ├── keys
│   ├── mmx64.efi
│   ├── refind.conf
│   ├── shimx64.efi
│   └── vars
└── tools
    ├── emptty.conf
    ├── motd
    └── seb.jpg

```
sbverify -l /efi/EFI/BOOT/grubx64.efi
```
signature 1
image signature issuers:
 - /CN=SupremeSessions
image signature certificates:
 - subject: /CN=SupremeSessions
   issuer:  /CN=SupremeSessions
```

objcopy --only-section .sbat -O binary /efi/EFI/BOOT/grubx64.efi /dev/stdout
```
sbat,1,SBAT Version,sbat,1,https://github.com/rhboot/shim/blob/main/SBAT.md
limine,1,Mintski,limine,9.3.0,https://www.github.com/limine-bootloader
```

mintsuki, 2025-06-25:
Hi,

Limine officially doesn't support nor endorse the use of `shim`, as it is a very poor attempt at circumventing the security granted by Secure Boot, and realistically there is no use case that warrants using it.

The reasonable way to use Secure Boot to secure a system with Limine is to sign Limine (after enrolling a config file hash using `limine enroll-config`).

closed 2025-06-25 by mintsuki

vuongtuha, 2025-06-26:
Okay, understand.

BinToss, 2025-11-02:
@mintsuki 
> The reasonable way to use Secure Boot to secure a system that uses Limine is to sign Limine using a custom key that is _manually added to the firmware's keychain_ (after enrolling a config file hash using limine enroll-config).

Adding keys to the firmware keychain is only feasible when Secure Boot Setup Mode is present in the UEFI UI. My RZ09-0423x (Razer Blade 17 (2022)), for instance, does not have this option. It should be possible to modify the firmware image to add the option, but that's way outside my area of expertise.