Limine-Bootloader/Limine #519: Support for loading drop-in drivers (/EFI/limine/drivers/*.efi)

JamiKettunen, 2025-07-11:
My [Asus Vivobook S15 (S5507QAD) laptop](https://www.asus.com/laptops/for-home/vivobook/asus-vivobook-s-15-s5507/) with 32GB of RAM and Qualcomm Snapdragon X Elite CPU requires devicetrees for booting due to important parts of ACPI power management being split to [Windows-specific PEP drivers](https://learn.microsoft.com/en-us/windows-hardware/drivers/kernel/using-peps-for-acpi-services); while annoying on mainlined and supported hw it's at least ok to deal with thanks to https://github.com/TravMurav/dtbloader working as an EFI driver being able to parse the DMI info and providing the bootloader with a devicetree so that "things just work" without requiring any `dtb_path` to be specified in the `protocol: linux` boot entries.

However, currently with Limine 9.4.0 this requires the manual installation of dtbloader by copying it to ESP + `efibootmgr --driver` command as per the project README to register it to be run on subsequent boots: there doesn't seem to be a way to automatically make it load from a path similar to `/EFI/systemd/drivers/dtbloaderaa64.efi` from ESP (for example `/EFI/limine/drivers/dtbloaderaa64.efi`?) as seen with systemd-boot which I've been using successfully so far. with https://github.com/limine-bootloader/limine/blob/v9.x/common/lib/bli.c around `bootctl` actually even reports the missing `Load drop-in drivers` support when I boot via Limine:
[image]

While for installed systemd or those which are bootstrapped from other existing Linux systems this isn't really a big deal it'd be really nice to be able to add EFI drivers to be included with e.g. aarch64 https://repo.chimera-linux.org/live/latest LIVE isos and have them automatically load so it's a "just works" experience for end users :)

lzcunt, 2025-07-13:
Not sure if automatically loading drop-in drivers matches Limine's current security model. Maybe a configuration key to load drivers would be a better idea, tho I am not sure if that would be pretty or better than the current situation. We can load drop-in drivers when not on secure boot tho.

Side note: the LoaderFeatures bit that bootctl reads for this feature is not documented so it's impossible to make that red X turn into a green tick without looking at systemd-boot code unfortunately. no green tick :( even if this is to be implemented

lzcunt, 2025-07-13:
I'm not sure how loading EFI drivers work but if it works like LoadImage/StartImage then yes signatures would be checked by the firmware. I'm specifically refraining from looking at systemd-boot because it's GPL and incompatible with Limine's licensing. `security_policy_install()` is probably shim's MOK stuff? I think using the firmware and shim to validate these drivers is a good compromise even though it doesn't really match Limine's baked hash approach. This might also allow Limine to load drop-ins automatically without a configuration key. I'm willing to implement this but I'll wait for Mint's response.

JamiKettunen, 2025-08-02:
@mintsuki Should the above be discussed more or would it be ok to implement? I've not yet setup secureboot for comparison with systemd-boot but I can if that'd be useful in any way

qookei, 2025-08-31:
Development has moved to Codeberg: https://codeberg.org/Limine/Limine. Please reopen this issue there.

closed 2025-08-31 by qookei