Limine-Bootloader/Limine #520: Feature Request: Reboot to ...

dazoe, 2025-07-12:
If your config has remember_last_entry: yes it should be trivial to add an option to the limine command to change the saved entry to allow a reboot to ... feature. I currently have it working using a shell script on linux but it would be nice is there was something like `#limine set-entry <#>`

```shell
#!/bin/bash

OPT="01"

case $1 in
        "c" | "console")
                OPT="02"
                ;;
        "g" | "graphic")
                OPT="01"
                ;;
        "w" | "windows")
                OPT="00"
                ;;
        *)
                echo "Missing option: console, graphic, windows"
                exit
                ;;
esac

echo "BOOT: $OPT"
chattr -i /sys/firmware/efi/efivars/LimineLastBootedEntry-513ee0d0-6e43-cb05-b272-f146a2fcb88a
printf "\x07\x00\x00\x00\x${OPT}\x00\x00\x00\x00\x00\x00\x00" > /sys/firmware/efi/efivars/LimineLastBootedEntry-513ee0d0-6e43-cb05-b272-f146a2fcb88a
chattr +i /sys/firmware/efi/efivars/LimineLastBootedEntry-513ee0d0-6e43-cb05-b272-f146a2fcb88a
```

lzcunt, 2025-07-13:
It might be cool to implement the necessary BLI variables to support `bootctl set-default`, `bootctl set-oneshot`, `systemctl reboot --boot-loader-entry=`, `systemctl reboot --boot-loader-menu=`, and `systemctl reboot --firmware-setup`.

I suggest against writing LimineLastBootedEntry for one-shot "reboot to ...". It will persist across reboots and might wear the NVRAM (probably negligible). Something like LimineBootOneShot would be nicer

lzcunt, 2025-07-13:
> Using my script to set the boot option to windows before rebooting from linux lets me just reboot I don't need to wait for the boot menu. It doesn't cause any extra writes to the nvram because those writes would happen anyway, it just so happened to happen before the reboot rather than at the boot menu.

I see

> I could change things up to using a script to modify the config file directly, I think that's how the bootctl set-default, and systemctl boot-loader options work, not 100% sure on that.

It uses persistent Boot Loader Interface EFI variables that are documented so we can implement the same interface, it's basically the same thing as LimineLastBootedEntry but with a nice userspace interface.

> If nvram wear is a truly a concern then the whole remember last entry feature should be changed to save the options on disk rather than in the nvram.

It's probs negligible unless your NVRAM is already dead.

> If remember_last_entry was changed to save on disk then that would allow bios support as well.

That would require write support for FAT

qookei, 2025-08-31:
Development has moved to Codeberg: https://codeberg.org/Limine/Limine. Please reopen this issue there.

closed 2025-08-31 by qookei