Limine-Bootloader/Limine #530: Limine hangs while loading the initrd on Dell Latitude E6220

mrunix00, 2025-08-16:
When I try to boot into any operating system that uses Limine's boot protocol and uses kernel modules/initrd, the bootloader hangs while loading the module.
I was only able to reproduce this issue on Dell Latitude E6220 laptop on UEFI boot (but not on BIOS) on Limine v9.6.1 (same issue on versions as old as 7.x).

mintsuki, 2025-08-17:
Hi. Thanks for reporting the issue. Could you provide more info? Like how big the modules you're trying to load are? How many modules?

It is a concern that your machine could just have a very slow firmware driver in UEFI mode for loading blocks from disk, and thus it appears like it hangs but it's actually just loading very slowly.

mrunix00, 2025-08-17:
> Hi. Thanks for reporting the issue. Could you provide more info? Like how big the modules you're trying to load are? How many modules?
>
> It is a concern that your machine could just have a very slow [...]
I only have 1 module, and it's 110K in size. How long should I expect it to load on the slowest UEFI firmware?

mintsuki, 2025-08-17:
Could you please try this version with added debug prints, and either copy the output here or post a picture. (It's not actually a .gz, GitHub needs that extension or else it won't upload; just remove it)

[BOOTX64.EFI.gz](https://github.com/user-attachments/files/21819783/BOOTX64.EFI.gz)

mrunix00, 2025-08-17:
> Could you please try this version with added debug prints, and either copy the output here or post a picture? (It's not actually a .gz, GitHub needs that extension or else it won't upload; just remove it)
> 
> [BOOTX64.EFI.gz](https://github.com/user-attachments/files/21819783/BOOTX64.EFI.gz)

Here's the output:
[image]

mintsuki, 2025-08-17:
Thanks, could you try this one now?

[BOOTX64.EFI.gz](https://github.com/user-attachments/files/21821793/BOOTX64.EFI.gz)

mrunix00, 2025-08-17:
> Thanks, could you try this one now?
> 
>

 [BOOTX64.EFI.gz](https://github.com/user-attachments/files/21821793/BOOTX64.EFI.gz)
[image]

mrunix00, 2025-08-17:
> Thanks, try this one next please.
> 
> [BOOTX64.EFI.gz](https://github.com/user-attachments/files/21821994/BOOTX64.EFI.gz)

mrunix00, 2025-08-17:
> Thanks, try this one next please.
> 
> [BOOTX64.EFI.gz](https://github.com/user-attachments/files/21821994/BOOTX64.EFI.gz)

[image]

mintsuki, 2025-08-17:
Thanks. Likewise.

[BOOTX64.EFI.gz](https://github.com/user-attachments/files/21822115/BOOTX64.EFI.gz)

mrunix00, 2025-08-17:
> It seems like this firmware is bugged and doesn't actually map memory above 4GiB as the spec mandates.

Hmm... Anything I can help with? You need any specific information or further testing?

mintsuki, 2025-08-17:
I'll close this issue for now as not planned. If you have further information that can show that this is not the firmware being broken, but a Limine issue instead, please re-open.

mintsuki, 2025-08-17:
If you can compile Limine yourself, you can try replacing the `!kernel` parameter on line 383 of file `common/protos/limine.c` with `false`. That will force load modules below 4GiB.

mrunix00, 2025-09-21:
> [@mrunix00](https://github.com/mrunix00) could you please try the latest trunk to see if it improves the situation?

I apologize for this, but I'm not home and I don't have that laptop with me, and there's a good chance that I won't be back home for months, so I won't be able to test it soon

mrunix00, 2025-10-05:
> [@mrunix00](https://github.com/mrunix00) could you please try the latest trunk to see if it improves the situation?

I just tested with Limine 10.0.1, that didn't fix it

mrunix00, 2025-10-06:
> > I was only able to reproduce this issue on Dell Latitude E6220 laptop on UEFI boot
> 
> Did you update to [the latest firmware](https://www.dell.com/support/product-details/en-uk/product/latitude-e6220/drivers) for this device?
> 

Yes

> You wrote:
> > When I try to boot into any operating system that uses Limine's boot protocol and uses kernel modules/initrd, the bootloader hangs while loading the module.
> 
> and
> 
> >  other operating systems like Windows or Linux work fine on this laptop (even on UEFI boot)
> 
> If you mean that Windows boots fine on this device using Limine `protocol: efi`, but when you try `protocol: limine` it fails, is that correct?
> 
> 

That's not what I meant, I meant that other operating systems/bootloaders work well and that my laptop does not have a broken firmware or something